citbrains/GankenKun_webots #9: protoモデルに関する作業

yasuohayashibara, 2021-04-18:
protoモデルに対する作業を共有するためのisseueです．
自由に追加していただいて構いません．

yasuohayashibara, 2021-04-19:
- [ ] パーツごとの質量の修正
- [ ] パーツごとの慣性行列の修正
あまり厳密でなくても良いと思います．厳密さを求めるとシミュレータが動作しなくなることがあります．
- [ ] モータの特性の追加
最大トルク，最大速度，粘性摩擦係数，クーロン摩擦（？），エンコーダの分解能
- [ ] バックラッシの追加
必ずしも使用するとは限らないので，GUIでON/OFFできるようにしたほうが良い．
- [ ] センサーの特性の入力
- [ ] 頂点数のチェック (50,000以下)
- [ ] 姿勢ファイルの作成
postures.json
- [ ] チームカラーやゼッケンの変更手順を指定されたものに変更
- [ ] レギュレーションを満たしているかのチェック

レギュレーション
• 5 <= BMI <= 30
• (FootWidth*FootHeight) <= (2.2*Hcom)ˆ2/32
• 1.2 <= max(FootWidth,FootHeight)/min(FootWidth, FootHeight) <= 3.5
• Width <= 0.55 Htop
• 0.35 Htop <= Hleg <= 0.7 Htop
• 0.1 Htop <= Hhead <= 0.3 Htop
• Only allowed sensors/actuators are used

yasuohayashibara, 2021-04-20:
モータのパラメータを導出するための計算式です．

[210420_モータのパラメータ.xlsx](https://github.com/citbrains/GankenKun_webots/files/6342080/210420_.xlsx)

AD58-3104, 2021-04-22:
HingeJointをHingeJointwithBacklashに変更した所、それら二つでは設定すべきパーツの関係が違った。その為SelfCollisionの判定で殆どのパーツ同士が干渉してしまい正常に立つ事が不可能になった。
パーツの関係の記述方法を全てHingeJointwithBacklashに対応したものに変更する必要がある。

yasuohayashibara, 2021-04-23:
こちらは今回間に合わなければこのままで良いと思いますが，パーツの命名規則がありますので，変更することが必要です．

v-hsc_model_specification_v1.0より
https://cdn.robocup.org/hl/wp/2021/04/v-hsc_model_specification_v1.0.pdf

```
Each robot needs to comport body parts with the following annotations as suffixes on the Solid names:
• [foot]: for all the body parts that can be in contact with the ground when walking (e.g. Solid.name "left
foot [foot]").
• [arm]: for all body parts between shoulder and hand. Those parts are not allowed to touch the ball
(e.g. Solid.name "left elbow [arm]").
Some joints (i.e., motors) should also be annotated with the following suffixes:
• [shoulder]: axis of the first joint of the arm (e.g. name "right shoulder pitch [shoulder]").
• [hip]: for the first leg joint with an axis lying parallel to the ground plane (e.g. name "left hip roll
[hip]").
```

yasuohayashibara, 2021-04-23:
- 以下の対処
https://github.com/citbrains/GankenKun_webots/pull/20#issuecomment-825308036

yasuohayashibara, 2021-04-23:
massとinertiaMatrixであるが，DARwIn-OPのGyroなどはphysics自身が無いので，それに倣って実装すれば良さそうである． 

https://github.com/cyberbotics/webots/blob/1367ab2560688c435afa50b8713522b9897afe7c/projects/robots/robotis/darwin-op/protos/Darwin-op.proto#L159-L166

yasuohayashibara, 2021-04-23:
慣性行列に正しい値を入れたところ振動的になり歩行できなくなった．
また，関節自身が外れる現象が見られるようになった．
あと，速度が1/10になっている．
https://youtu.be/TaZbklhiV1M

backlashの機能は実装中のようであるので，色々とバグがあるかもしれない．

yasuohayashibara, 2021-04-23:
以下命名規則に基づいてsolidの名称を変更した．
https://github.com/citbrains/GankenKun_webots/issues/9#issuecomment-825290685

yasuohayashibara, 2021-04-23:
歩行できなくなったのは残念ですが，protoファイルで行うことはほぼ行ったので，これを提出で良いと思います．
今回は歩行を行うことは前提条件に入っていないと思いますので，指示された内容をできるだけ満たしたprotoファイルを出したほうが良いと思います．
その点では，かなり完成度は高いと思います．

yasuohayashibara, 2021-04-23:
慣性行列の影響を示す動画です．
https://youtu.be/pBFOf7Vjj10

大きくすることで，回転しづらくなり歩行が安定します．

yasuohayashibara, 2021-04-23:
backlashを外したところふらつきはするが歩行できるようになった．
**モデル自身には致命的な問題は無い**と思われる．

backlashの開発により，歩行できない問題は自然と解決するのではないかと考える．
（どちらにしても計算時間が長いので，現状では試合に使用するのは難しいと思います．）

現在のprotoファイルですが，慣性行列を正しい値にすることで，安定しすぎて不自然であったのが実際の動きに近づいたと思います．

変更内容　HingeJointWithBacklash -> HingeJoint

動画
https://youtu.be/w7YXdusKpGs

AD58-3104, 2021-04-23:
backlashに関してはif文でON/OFFをGUIで変更できるようにしている実装がdarwin-opではされているので、それを参考にGankenKunでも同じものを実装しておこうと思います。

yasuohayashibara, 2021-04-23:
現在のモデルですが，RoboCupのフィールドでは歩行することを確認しました．
ただし，遅いですが．．．
https://youtu.be/ik0Vxq4sZUg

環境によってこんなにも動きに差が出るのですね．．．
何が要因かを調べたい気もしますが，とりあえず動くので良しとしたいと思います．

yasuohayashibara, 2021-04-23:
どうもworldファイルの以下のパラメータが関連しているようです．
以下の値のときだけ正常に動作します．

basicTimeStep 8

AD58-3104, 2021-04-25:
となるとワールドのTimeStepは8で固定するのが良さそうですね。

yasuohayashibara, 2021-04-25:
#22 で示したように，設定ファイルに基づいて色とゼッケンの変更ができるようになったため，protoモデルに対する最低限の作業は終了したと思います．
issueを閉じます．

closed 2021-04-25 by yasuohayashibara