citbrains/GankenKun_webots #10: proto編集箇所報告とメモ

AD58-3104, 2021-04-19:
proto作業用です。

AD58-3104, 2021-04-19:
実際のファイル変更作業を複数人で平行して行うのはほぼ無理な感じになったのでここで変更箇所をまとめて変更作業は一人がやる事にします。

AD58-3104, 2021-04-20:
gearの暫定的なbacklashは0.01。

yasuohayashibara, 2021-04-20:
0.01rad -> 0.5729 deg
もしかしたら計測することを要求されるかもしれませんが妥当だと思います．

AD58-3104, 2021-04-20:
モータのエンコーダーの分解能は12bitで0.0015

AD58-3104, 2021-04-20:
MPU9250 
加速度計+-16[G] を[m/ss]に変換し設定
ジャイロ +-2000[degree/s]を[rad/s]に変換し設定

AD58-3104, 2021-04-20:
変換後
加速度計 156.91
ジャイロ 34.91

yasuohayashibara, 2021-04-20:
> MPU9250
> 加速度計+-16[G] を[m/ss]に変換し設定
> ジャイロ +-2000[degree/s]を[rad/s]に変換し設定

MPU9250はレンジを調整して使用しています．
（まあそこまでこだわらなくて良いと思いますが念のため）
加速度±8G
ジャイロ±1000deg/s

以下に設定があります．
https://github.com/citbrains/citbrains_humanoid/blob/master/for2050/src/hr46_b3m/RTIMULib.ini

AD58-3104, 2021-04-20:
慣性モーメントは久保寺くんが修正してくれたモデルで重心周りの慣性モーメントを見る。
Ixx Ixy IxzのベクトルとIxx Iyy Izzのベクトルを入力していく。inventorの数値がg mm^2なのでkg m^2であるprotoには1000で割った数値を入力。

AD58-3104, 2021-04-20:
posture.jsonはupright は全ての関節角を0,extensionを左右の肩ロールを1/2rad,他を全て0のもので良さそう。

AD58-3104, 2021-04-20:
チームカラーはサンプルプログラム等が出るまでは取り敢えず放置の方針でいく

AD58-3104, 2021-04-20:
頂点数
合計しても約15000ほどである為問題なさそう。

MasatoKubotera, 2021-04-20:
下記のスプレッドシートにまとめました．
[210420_部位ごとの質量・質量モーメント](https://docs.google.com/spreadsheets/d/1A5OjJvEKwI21zuCdGR1n0Xrg8VCY3uxQGpSIrZuM5nY/edit#gid=0)

yasuohayashibara, 2021-04-20:
頂点数ですが，もっと少なくなると思います．
B

[image]

yasuohayashibara, 2021-04-20:
入江さんに出力していただいたポリコン数をこちらにも載せておきます．

```
python -m urdf2webots.importer --input=gankenkun.urdf
Robot name: gankenkun
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/body.stl
1052
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/head.stl
372
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/shoulder.stl
102
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/arm.stl
476
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/hand.stl
128
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/waist_yaw.stl
94
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/waist_roll_left.stl
542
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/slider_crank.stl
56
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/waist_pitch.stl
124
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/waist_pitch_mimic.stl
108
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/knee_pitch_left.stl
364
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/shin_pitch.stl
124
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/shin_pitch_mimic.stl
156
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/independent_pitch.stl
136
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/ankle_pitch_left.stl
324
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/ankle_roll_left.stl
76
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/waist_roll_right.stl
542
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/knee_pitch_right.stl
364
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/ankle_pitch_right.stl
324
Parsing Mesh: /home/irie/GankenKun_pybullet/URDF/URDF/meshes/STL/ankle_roll_right.stl
76
```

yasuohayashibara, 2021-04-21:
> 慣性モーメントは久保寺くんが修正してくれたモデルで重心周りの慣性モーメントを見る。
> Ixy Ixz IyzのベクトルとIxx Iyy Izzのベクトルを入力していく。inventorの数値がg mm^2なのでkg m^2であるprotoには10^9で割った数値を入力。

慣性行列の入力で問題となるのは軸の向きで，多関節ロボットは非常に分かりづらいです．
ジョシュア先輩の作成したURDFファイルを参考にすると良いと思います．

bodyの例
https://github.com/citbrains/opr_ros/blob/kinetic-devel/opr_description/urdf/version3/body/body.urdf.xacro

数字が明らかに異なっているところは，xyzの軸の向きが入れ替わっているおそれがあります．

こちらの問題は以下でも議論しています．
https://github.com/citbrains/GankenKun_pybullet/issues/9
※ ROSとrvizを使用するとxyzの向きを確認することができます．

AD58-3104, 2021-04-21:
@yasuohayashibara 
LUTは以下の様に設定しました。
          Accelerometer {
            name "imu/data accelerometer"
            lookuoTable [-156.91 0 0,
                         156.91 65536 0 ]
          }
          Gyro {
            name "imu/data gyro"
            lookuoTable [-34.91 0 0,
                         34.91 65536 0 ]
          }
こちらについて適切かどうかが自分では判断しかねるので確認して頂けないでしょうか。
webotsでのLUTについての資料です。
https://cyberbotics.com/doc/reference/distancesensor?tab-language=c++#lookup-table

yasuohayashibara, 2021-04-21:
@AD58-3104 
承知しました．

hiroki-0001, 2021-04-21:
> モータの特性の追加
最大トルク，最大速度，粘性摩擦係数，クーロン摩擦（？），エンコーダの分解能 

モータ特性に関して、調べ終わったので以下に記載しておきます。 

|     |  B3M-SC-1040-A   |  B3M-SC-1170-A   | 
| --- | --- | --- | 
|  最大トルク[Nm]   |  4.6   |  7.6   | 
|  最高速度[rpm]   |  54   |   46  | 
|  粘性摩擦係数[rad/s/(Nm)]   |  1.2452588169   |  0.639974   | 
|  クーロン摩擦[Nm]  |  0.05888   |  0.07296   |

エンコーダの分解能は上記に記載されているので省略します。

また静摩擦力の計算の際に使用したデューティ比を求める際に使用した2つの値(PWM Frequency = 周波数, PWM Duty = カウント数)は以下の通りです。

|     |  B3M-SC-1040-A   |  B3M-SC-1170-A   | 
| --- | --- | --- | 
|   PWM Frequency[Hz]  |  20000   |  20000   | 
|  PWM Duty(カウント数)   |  32   |   24  | 

PWM Frequencyの値が各機体や関節部位ごと異なっていましたが上記の計算には、現在もっとも動きがいい167(新機体)のPWM Frequencyの最頻値である20000Hzを使用しました。

おそらくですが機体ごとにPWM Frequencyはバラバラに設定されていると考えられるので、この件に関してはcitbrains_humanoidのほうにissueを立てて対応します。

hiroki-0001, 2021-04-21:
[image]

基板部分の重さを計量しました。
スイッチ部分も含めると250gほどでした。

kiyoshiiriemon, 2021-04-21:
@AD58-3104 
@yasuohayashibara

> LUTは以下の様に設定しました。
> Accelerometer {
> name "imu/data accelerometer"
> lookupTable [-156.91 0 0,
> 156.91 65536 0 ]
> }
> Gyro {
> name "imu/data gyro"
> lookupTable [-34.91 0 0,
> 34.91 65536 0 ]
> }
> こちらについて適切かどうかが自分では判断しかねるので確認して頂けないでしょうか。
> webotsでのLUTについての資料です。
> https://cyberbotics.com/doc/reference/distancesensor?tab-language=c++#lookup-table

最大値を65536にしても、16ビット整数で返してくれるわけではないですし、最大レンジをソフトウェアで設定できたりもしないので、ここは物理単位そのままにした方が汎用性があってよいと思います。
APIもrad/s, m/s^2のインタフェースですよね。

AD58-3104, 2021-04-22:
@kiyoshiiriemon 
ありがとうございます。物理単位に変更しました。

AD58-3104, 2021-04-22:
足先を芝生に埋もれさせない様に足のboundingobjectのtransformを1cm下に下げた

hiroki-0001, 2021-04-22:
慣性行列追加と質量の値の修正の作業の際に以下の問題があったので共有します。

- "left_slider_crank_link", "right_slider_crank_link" 　ここの部分は長穴機構なのでトルクを2倍、速度を半分にすべき？(変更する場合はモータのパラメータを変更したほうがいい)

- "left_foot_link", "right_foot_link"の2つがコード上には存在するが、webotsのGUI上のどこにあるかが不明

yasuohayashibara, 2021-04-22:
申し訳ありませんが，self-collision対策のためにmainのprotoファイルを大幅に変更しました．
マージしていただくことはできますでしょうか．

AD58-3104, 2021-04-22:
現在fix_proto_for_competitionブランチで開発しています。先にmainの変更をこちらのブランチに取り込み、動作を確認します。

yasuohayashibara, 2021-04-24:
protoファイルを提出しましたので，こちらのissueを閉じます．
さらなる改良は別のissueで議論したいと思います．

closed 2021-04-24 by yasuohayashibara