citbrains/GankenKun_webots #15: 明日(04/23)までに必要なデータの確認と完成

yasuohayashibara, 2021-04-21:
明日(04/23)がモデルファイル提出の期限となります．
提出するドキュメントは以下のように複数（最低４つ）あります．
（林原もまだ十分に確認できていません．）

|-- documentation
| |-- documentation.pdf
| |-- datasheet1.pdf
| |-- ...
|-- RobotName1
| |-- RobotName1.proto
| |-- postures.json
| |-- ...

ミーティングでドキュメントは書きましょうかという話をしたような気もしますので，担当者がいなければ林原が今晩雛形を作成したいと思いますが，モデルファイルが完成していないと書きづらいです．

モデルファイルの変更はできれば本日中(04/22)に行っていただければ助かります．
なお，変更した場合，動作チェックしてからmasterにマージしてください．
（webotsで安定して歩行できるかで判断してください）
masterにマージしていないものはドキュメントに反映できませんので，ご注意ください．

また，明日ギリギリで変更するとチェック不足で動作しないモデルを出すおそれが高くなりますので，主な変更は本日までにしたほうが良いと思います．

明日はチェックと微修正のみにした方が良いと思います．

AD58-3104, 2021-04-22:
設定すべき項目はロボットのカラー選択を除き(こちらは次回提出の際に修正すれば良いと思われます)設定しましたが、シミュレーション開始時に確実に転倒するようになってしまいました。その後GUIで操作して立たせると歩行自体は可能なので

yasuohayashibara, 2021-04-22:
現在のmainのprotoファイルは干渉しなくしています．
#18 を参照

yasuohayashibara, 2021-04-23:
API Specifications for Virtual Soccer Competition
のドキュメントも提出しなければならないそうです．

https://cdn.robocup.org/hl/wp/2021/04/v-hsc_simulator_api_draft2.pdf

こちら桑野君が作成してくれるとのことです．

AD58-3104, 2021-04-23:
@RikuYokoo 
カメラの画角関連のprotoの編集が終わったらmainブランチにコミットしておいて下さい。

GakuKuwano, 2021-04-23:
API Specifications for Virtual Soccer Competition内のConfiguration file (team.json)の作成は完了しました．

GakuKuwano, 2021-04-23:
@yasuohayashibara @AD58-3104 
以下，提出用のrobot modelです．確認お願いします．
[CITBrains.zip](https://github.com/citbrains/GankenKun_webots/files/6365786/CITBrains.zip)

GakuKuwano, 2021-04-23:
@yasuohayashibara @AD58-3104 
以下，提出用のrobot modelのzipとConfiguration file (team.json)のzipです．
確認お願いします．
[CITBrains.zip](https://github.com/citbrains/GankenKun_webots/files/6365786/CITBrains.zip)

[team_json.zip](https://github.com/citbrains/GankenKun_webots/files/6365828/team_json.zip)

GakuKuwano, 2021-04-23:
4/23までに必要なものは提出完了しました．

yasuohayashibara, 2021-04-24:
完了したので，issueを閉じます．

closed 2021-04-24 by yasuohayashibara