citbrains/GankenKun_webots #16: 部品のデータシート（英文）の収集

yasuohayashibara, 2021-04-22:
protoファイルに関連する部品のデータシート（できるだけ英文）を送ることが必要となる．
集めてアップロードしていただければと思います．
とりあえずこちらのissueに添付で良いと思います．
（もしくはGoogle Driveのアドレスを誰か指定してください．）

Robot Model Specifications v1.0 より
https://cdn.robocup.org/hl/wp/2021/04/v-hsc_model_specification_v1.0.pdf

訳
ロボットモデルの他に、例年のロボット仕様書に似た、より充実したドキュメントを提供する必要があります。このドキュメントには、ロボットに使用されているセンサー（モーター、加速度計、ジャイロスコープ、カメラなど）の関連データシートと、データシートからプロトのパラメータをどのように導き出したかの説明が含まれていなければなりません。

yasuohayashibara, 2021-04-22:
収集したデータシート
※追加していただいて構いません．

- [ ] モータ
- [ ] IMU
- [ ] カメラ

hiroki-0001, 2021-04-22:
B3M-SC-1040-A, B3M-SC-1170-Aのデータシートの英語版を見つけたので、ここにあげておきます。
[B3M_SoftwareManual1.2.0.3_en.pdf](https://github.com/citbrains/GankenKun_webots/files/6356784/B3M_SoftwareManual1.2.0.3_en.pdf)

KANBE8810, 2021-04-22:
KBCR-S02MUのデータシートなのですが、外国為替および外国貿易管理法により輸出または海外への提供が規制されているそうで、その影響かデータシートがありませんでした。
一応、英語で書かれているPDFとサイトがありましたので記載しておきます。
こちらの2ページ目にKBCR-S02MUの記載がありました。
[Brochure_CameraAll_V4.40.pdf](https://github.com/citbrains/GankenKun_webots/files/6356782/Brochure_CameraAll_V4.40.pdf)
英語版サイト
https://www.shikino.co.jp/eng/products/product-kbcr-s02mu_1.php
また、日本語版ではありますが詳しく書かれているものがあったので共有しておきます。
[KBCR-S02MU_Ver1.01_J.pdf](https://github.com/citbrains/GankenKun_webots/files/6356840/KBCR-S02MU_Ver1.01_J.pdf)

ueshou, 2021-04-22:
MPU-9250 データシートです。見つけた中で、上から古い順です。一番下が見つけた中では最新です。
↓ MPU-9250 Register Map and Descriptions Revision 1.4
https://strawberry-linux.com/pub/RM-MPU-9250A-00.pdf
↓ MPU-9250 Product Specification Revision 1.0　
https://strawberry-linux.com/pub/PS-MPU-9250A-01.pdf
↓ MPU-9250 Product Specification Revision 1.1
https://product.tdk.com/system/files/dam/doc/product/sensor/mortion-inertial/imu/data_sheet/ps-mpu-9250a-01-v1.1.pdf

yasuohayashibara, 2021-04-22:
ご協力ありがとうございます．

yasuohayashibara, 2021-04-22:
datasheetを以下にアップロードしました．
https://drive.google.com/drive/folders/1T8wtw6NVLPWkKKW7HNVonhCTgqkYIPNV?usp=sharing

issueを閉じます．

closed 2021-04-22 by yasuohayashibara