citbrains/GankenKun_webots #21: webotsでRoboCupの環境を実行するまでの手順

yasuohayashibara, 2021-04-24:
以下にwebotsでRoboCupの環境を実行するまでの手順が書かれているが，省かれているところがあり，Linuxに慣れていない人には難しいかもしれないので手順を示す．
https://github.com/cyberbotics/webots/tree/feature-robocup-controllers/projects/samples/contests/robocup#readme

連休中に取り組んで欲しいので，連休前にドキュメントを仕上げる．

yasuohayashibara, 2021-04-24:
実行するまでに問題となったのは，主に２つ

- protobuf
- GameController

すでにソフトウェア開発の環境がある人には問題とならないかもしれないが，セットアップから行うことが必要になるので手順を残しておく．

kiyoshiiriemon, 2021-04-24:
とりあえず、webotsをソースからfeature-robocup-controllersブランチでビルド・動作できることを確認しました。

* 準備
```
git clone --recursive -b feature-robocup-controllers https://github.com/cyberbotics/webots
cd webots
git branch (ブランチfeature-robocup-controllersにいることを確認する)
sudo ./scripts/install/linux_compilation_dependencies.sh 
sudo apt install libglm-dev
```
もしcloneするときに--recursiveをつけ忘れたら ```git submodule update --init --recursive``` を実行する

* コンパイル
    -  make -j4   (jの後の数字はCPUコア数にするとよい)

* トラブルシューティング
   - makeが途中で進まなくなることがよくある。原因はWebotsのサーバーが弱いのか、ダウンロードの失敗
   - 何度かやり直してもうまく行かない場合は、ファイルを直接配置するとよい [ここからダウンロードできるようにしました](https://drive.google.com/drive/folders/1eSE1_oEx44hJDY8_laZ3aRTnZR8ETxT3?usp=sharing)
   - dependencies/ に配置
       - webots-qt-5.15.0-linux64-release.tar.bz2
       - lua-gd-linux64.tar.bz2
       - libassimp.5.0.1.tar.bz2
       - libOIS.1.4.tar.bz2
       - libpico.tar.bz2
  - projects/default/resources/ に配置
       - sumo-0.30-linux64.tar.gz

yasuohayashibara, 2021-04-25:
以下にwebotsのインストール方法がある．
https://github.com/cyberbotics/webots/wiki/Linux-installation

yasuohayashibara, 2021-04-25:
以下のwikiに転載したので，issueを閉じます．
https://github.com/citbrains/GankenKun_webots/wiki/webots%E3%81%A7%E8%A9%A6%E5%90%88%E3%81%AE%E7%92%B0%E5%A2%83%E3%82%92%E4%BD%9C%E6%88%90%E3%81%99%E3%82%8B%E6%89%8B%E9%A0%86

closed 2021-04-25 by yasuohayashibara