citbrains/GankenKun_webots #34: TC検証(05/23)までにDockerイメージの作成

KANBE8810, 2021-05-05:
Dockerイメージを5月23日までに4機体分を作成して提出する。
提出するTC検証用のDockerイメージは、技術委員会がリアルタイムファクタで、大会で使用されるハードウェア上でシュミレーションできる必要がある。

yasuohayashibara, 2021-05-05:
作成するDockerは以下の形で間違い有りませんでしょうか？
https://github.com/citbrains/GankenKun_webots/issues/28#issuecomment-828860729

KANBE8810, 2021-05-05:
おそらく間違いないとは思いますが、10日にAPIとかの仕様書の更新があるので、そちらの方で改めて確認したいと思います。

KANBE8810, 2021-05-13:
検証用のDockerfleが完成してimage buildとdocker runも通りました。
これからwebotsとの通信がうまくいくかどうかを確かめたいと思います。

Dockerfileには問題がなかったのですが、作成したimageが5.11GBとルールの5GBを超えてしまっているので対策が必要です(darknetとかもいれようとすると更にシビアになるかもしれません)
現状ではcitbrains_humanoidからcloneするものを減らすという形で対応したいと考えています。

yasuohayashibara, 2021-05-13:
dockerの仕組みを十分に理解していないのですが，実際の試合で提供するのはバイナリになると思いますので，ソースコードやgithubにある過去の履歴などは削除したdockerで検証して良いと思います．
dockerの中で開発するのであれば，ソースコードも含めた方が良いかもしれませんが．

kiyoshiiriemon, 2021-05-14:
ソースや中間オブジェクトは消すのがよいと思います。
またgit clone --depth=1 でクローンすれば最新の履歴だけ取得できるのでサイズを減らすことができます。

KANBE8810, 2021-05-14:
入江さんのコマンドを使用したところ5.11GBから4.57GBまで減らすことができました。
画像1枚目が普通にcloneしてmakeしたとき、
画像2枚目が入江さんのコマンドでcloneしてmakeしたときのdfの結果です。
Availableの値が大きく変わっていますが、これは僕の端末内の不要なimagesを削除しただけなので気にしないでください。

[image]
[image]

KANBE8810, 2021-05-14:
調べていたらマルチステージビルドをするとコンテナイメージを大幅に抑えられる可能性があるという記事を見つけたので、次はそちらを試してみたと思います。

KANBE8810, 2021-05-15:
runstrategy.pyの起動まで確認して、srcのmakeまで自動でやるようにdockerfileを書きました。

KANBE8810, 2021-05-18:
### 23日までにやること
- [ ] 歩行用コードの開発

- [ ] IPアドレスなどの設定

- [ ] リアルタイムでシュミレーションできるかを確認

### 6月の模擬試合までにやること
- [ ] YOLOに対応させる(NVIDIAとかをインストール)

上から順々にやっていこうと思います。
他に23日までにDocker関連でやらなければならないことがあれば指摘してください。

KANBE8810, 2021-05-20:
とりあえずその場歩きを永遠と行うコードを書いて、develop/RobVup2021で動作確認を行ったところ、問題なく歩行を1分以上続けられていました。
https://github.com/citbrains/citbrains_humanoid/blob/develop/RoboCup2021/for2050/src/pyfiles/endless_walk.py

yasuohayashibara, 2021-05-21:
入江さんのブランチが最終的に提出するプログラムとなると思いますので，そちらで検証していただければと思います．
（先ほどprotoモデルをアップデートしたので，関節の回転が逆方向になるかもしれません．）

yasuohayashibara, 2021-05-21:
本日(05/22)，本番の環境を用意しますので，そちらでチェックしましょう．
そういえばwebotsサーバの計算結果を表示する方法が分からないままですが．．．

KANBE8810, 2021-05-22:
複数コマンドを実行させなきゃいけないことを完全に忘れてました。
現状としてはdocker-composeでそれぞれを実行するコンテナを作成してコンテナ間通信で戦略を動かそうかと考えています。
何かもっと良い方法があれば教えていただきたいです。 @kiyoshiiriemon

KANBE8810, 2021-05-22:
解決しました。

yasuohayashibara, 2021-05-22:
フランクフルトのサーバーでwebotsの起動を確認しました．
まだポートを開けていないので，これから開けます．
必要な設定がありましたら，行いますのでよろしくおねがいします．

user: ubuntu
password: ubuntu
アドレス　3.122.251.119
（フランクフルトなので遅いです）

[image]

yasuohayashibara, 2021-05-22:
ポートを開けて以下のメッセージが戻ってくることを確認しました．

- webotsサーバを起動していない場合
```
mirai@mirai-virtual-machine:~/robocup/for2050/src/webots$ ./WebotsClient
Cannot connect to 3.122.251.119:10021
Connection closed by server.
```
- webotsサーバを起動した場合
```
mirai@mirai-virtual-machine:~/robocup/for2050/src/webots$ ./WebotsClient
Connection to 3.122.251.119:10021 refused: your IP address is not allowed in the game.json configuration file.
TIME=0
```

yasuohayashibara, 2021-05-26:
終了しましたのでissueを閉じます．

closed 2021-05-26 by yasuohayashibara