citbrains/GankenKun_webots #39: Dockerイメージのサーバーへのアップロード

yasuohayashibara, 2021-05-21:
Dockerイメージをサーバーにアップロードする手順を明確にする．

yasuohayashibara, 2021-05-21:
桑野君が受け取ったメッセージ
-----

Dear Gaku Kuwano,

attached you will find the personal credentials for team CIT Brains to
push Docker images to the AWS container registry for the RoboCup
Humanoid League Virtual Competition. Important: Do not share this
credentials with anyone outside your team or include this in any
publicly available files!

Instructions on how to push Docker images are provided here:
https://cdn.robocup.org/hl/wp/2021/05/v-hsc_server_specification_draft3b.pdf
Please note that this registry will be used for the second round of
robot model submissions to provide the simple walking behavior of your
robot, as well as for the mock-competition and the RoboCup world
championship. When pushing docker images, two important constraints
apply: (1) individual docker images may not exceed 5 GB. Docker files
violating this constraint will be deleted and not considered by us for
the competition. (2) You can use different tags for different versions
of your software. If you use the same tags for two images, the old
version will be overwritten. Please use this or remove images that are
no longer needed to save us hosting space and related costs during the
competition.

If you have any questions, don't hesitate to contact us on Discord or in
the forum.

Kind regards,
Maike (for the RoboCup Humanoid League Technical Committee)

--
Dr. Maike Paetzel-Prüsmann
Postdoctoral Researcher
Foundations of Computational Linguistics Laboratory
University of Potsdam, Germany

https://drive.google.com/file/d/13cmZ-cH9vu4w5qdrcPZqBbH8ENr6YdpU/view?usp=sharing

-----

Dear Gaku Kuwano,

チームCIT BrainsがロボカップのためにAWSコンテナレジストリにDockerイメージをプッシュするための個人認証情報が添付されています。
ロボカップ ヒューマノイドリーグ仮想大会のためにAWSコンテナレジストリにDockerイメージをプッシュするためのチームCIT Brainsの個人認証情報が添付されています。
ヒューマノイドリーグ仮想大会 重要です。この資格情報をチーム以外の人と共有しないでください。
重要：この認証情報をチーム以外の人と共有したり、一般に公開されているファイルに
公開されているファイルに含めないでください。

Dockerイメージをプッシュする方法については、こちらで説明しています。
https://cdn.robocup.org/hl/wp/2021/05/v-hsc_server_specification_draft3b.pdf
なお、このレジストリは、ロボットモデルの第2ラウンドの応募の際に
このレジストリは、ロボットの簡単な歩行動作を提供するために、ロボットモデルの第二次応募の際に使用されます。
このレジストリは、ロボットの簡単な歩行動作を提供するための第二次ロボットモデル提出時や、模擬競技会、ロボカップ世界大会で使用されます。
を使用します。dockerイメージをプッシュする際、2つの重要な制約があります。
が適用されます。(1) 個々のDockerイメージは5GBを超えてはならない。ドッカーファイル
この制約に違反したDockerファイルは削除され、大会では考慮されません。
考慮されません。(2)ソフトウェアのバージョンごとに異なるタグを使用することができます。
(2) ソフトウェアのバージョンごとに異なるタグを使用することができます。2つのイメージに同じタグを使用した場合、古いバージョンは上書きされます。
のバージョンが上書きされてしまいます。使用するか、不要になった画像を削除してください。
大会期間中のホスティングスペースや関連費用の節約のために、これを使用するか、不要になった画像を削除してください。
大会期間中のホスティングスペースと関連コストを節約するために、これを使用するか、不要になった画像を削除してください。

ご不明な点がありましたら、ご遠慮なくDiscordまたはフォーラムでお問い合わせください。
フォーラムまでご連絡ください。

よろしくお願いします。
マイケ（ロボカップ・ヒューマノイドリーグ技術委員会担当

--
マイケ・ペッツェル-プリュスマン博士
ポスドク研究員
ポツダム大学 計算言語学の基礎研究室
ポツダム大学（ドイツ

https://drive.google.com/file/d/13cmZ-cH9vu4w5qdrcPZqBbH8ENr6YdpU/view?usp=sharing

kiyoshiiriemon, 2021-05-22:
@yasuohayashibara @KANBE8810 
ここは公開なので認証情報は貼らない方がいいと思います。
自分のアカウントで実験しました。ざっとこんな手順でpushできるようです。

```
sudo apt  install awscli
aws configure
aws ecr get-login-password | docker login --username AWS --password-stdin https://<aws_account_id>.dkr.ecr.<region>.amazonaws.com
docker tag <作ったDockerイメージ> <aws_account_id>.dkr.ecr.<region>.amazonaws.com/<指定されたcitbrains用の名前>
docker push <aws_account_id>.dkr.ecr.<region>.amazonaws.com/<指定されたcitbrains用の名前>
```

<aws_account_id>は12桁の数字
<region>は指定されたregion (今回はus-east-2のようですね)

yasuohayashibara, 2021-05-22:
申し訳ありません．
公開ということを忘れておりました．

yasuohayashibara, 2021-05-23:
アップロードできましたので，issueを閉じます．
@KANBE8810 できましたら忘れないうちに手順を記録しておいていただければと思います．

closed 2021-05-23 by yasuohayashibara

KANBE8810, 2021-05-26:
pushの手段はサーバのドラフトに書いてある通りで、入江さんの手順でできると思います。