citbrains/GankenKun_webots #52: SupervisorによるGankenKunの各関節とボールの座標取得

KANBE8810, 2021-12-22:
姿勢推定の学習を回すためには、ロボットの各関節の座標を取得する必要があるため、WebotsのSupervisorを利用して座標を取得する。また、同時にボールの座標も取得する。
コードとしてはhr46_b3mのmain.cpp内に記載する予定です。

yasuohayashibara, 2021-12-22:
Pythonになりますが，以下のコマンドでボールの位置は取得できます．
ただし，まだボールは検出しなくても良いかも知れませんが．．．

https://github.com/citbrains/GankenKun_webots/blob/8657861f036734929801c638da6b0e4e2d2b5038/controllers/learning/learning.py#L73

yasuohayashibara, 2021-12-27:
リンク？位置を戻すサンプルを作りました．
想像していたのよりかなり大変でした．
一番大変なのはドキュメントがないところですが．．．

https://github.com/citbrains/GankenKun_webots/pull/53

KANBE8810, 2022-06-07:
https://github.com/citbrains/GankenKun_webots/pull/64 で開発された2次元座標を組み込もうとした所、キーポイントがうまく読み込めない問題が発生した。
原因はおそらく生成されたロボットにcolorの欄が無いために認識できないもと思われる。

branth : get_position
使用world : get_position_test.wbt
[image]

KANBE8810, 2022-08-23:
上記の問題は推定するロボットのカラーを赤色に変更したところ解決しました