citbrains/GankenKun_webots #78: agilerlのエラー

yasuohayashibara, 2023-11-27:
#77 を実行したら以下のエラーが発生した．
@GakuKuwano 何か心当たりがあれば教えてください．

```
0%| | 0/10000 [02:00<?, ?it/s]
Traceback (most recent call last):
  File "agilerl_train.py", line 225, in <module>
    agent.learn(experiences)  # Learn according to agent's RL algorithm
  File "/home/haya/GankenKun_webots_kuwano/controllers/agilerl_train/soccer/matd3.py", line 459, in learn
    input_combined = torch.cat(list(states.values()) + action_values, 1)
RuntimeError: Tensors must have same number of dimensions: got 2 and 3
DELETE
WARNING: 'agilerl_train' controller exited with status: 1.
```

yasuohayashibara, 2023-11-27:
ちなみに，開始時に以下のメッセージが流れます．
ロボットはしばらく動いていて，おそらく訓練のフェーズに入ったときに上記のメッセージを出力して停止します．

```
INFO: agilerl_train: Starting controller: python3 -u agilerl_train.py
/home/haya/GankenKun_webots_kuwano/env/lib/python3.8/site-packages/torch/cuda/__init__.py:138: UserWarning: CUDA initialization: The NVIDIA driver on your system is too old (found version 11040). Please update your GPU driver by downloading and installing a new version from the URL: http://www.nvidia.com/Download/index.aspx Alternatively, go to: https://pytorch.org to install a PyTorch version that has been compiled with your version of the CUDA driver. (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
===== AgileRL Online Multi-Agent Demo =====
WARNING: '/home/haya/GankenKun_webots_kuwano/protos/GankenKun_simple.proto': This file was created with Webots R2021a while you are using Webots R2021b. You may need to adjust urls for textures and meshes, see details in the change log of Webots R2021b.
```

GakuKuwano, 2023-11-28:
おそらく， NVIDIA driverのバージョンが古いのが原因かもしれません．
NVIDIA driverのバージョンを上げるか，PyTorchのバージョンを下げることで解決するかもしれませんが，後者の場合はPyTorchのバージョンによっては動作しない可能性が考えられます．

もしくは別の方法として，以下の箇所を` device = torch.device("cpu") `にして，GPUを使わずに学習アルゴリズムを回せば，バージョンを変えずに動作させることができるかもしれません．
https://github.com/citbrains/GankenKun_webots/blob/00dc6cf7a68b330dee242334417da75a30b6f776/controllers/agilerl_train/agilerl_train.py#L22

yasuohayashibara, 2023-12-05:
ドライバのバージョンのエラーは無くなったのですが，以下のエラーが引き続き発生します．
@GakuKuwano 後で相談させてください．

```
0%| | 0/10000 [01:13<?, ?it/s]
Traceback (most recent call last):
  File "agilerl_train.py", line 225, in <module>
    agent.learn(experiences)  # Learn according to agent's RL algorithm
  File "/home/haya/GankenKun_webots_kuwano/controllers/agilerl_train/soccer/matd3.py", line 459, in learn
    input_combined = torch.cat(list(states.values()) + action_values, 1)
RuntimeError: Tensors must have same number of dimensions: got 2 and 3
DELETE
WARNING: 'agilerl_train' controller exited with status: 1.
```

yasuohayashibara, 2023-12-05:
agilerlのインストールの仕方が異なっているかもしれません．
私はpipでインストールしましたが，現在学習中のものはソースからインストールしました？

GakuKuwano, 2023-12-05:
pipでインストールしていますが，agilerl内のソースコードに変更を加えるために，一部インポートの対象となるパスを変えています．

以下はpipでインストールしたパッケージの一覧です．
---
accelerate==0.18.0
agilerl==0.1.13
dill==0.3.7
fastrand==1.8.0
flatten_dict==0.4.2
gymnasium==0.28.1
h5py==3.9.0
hydra-core==1.3.2
matplotlib==3.5.3
minari==0.4.2
numpy==1.24.3
omegaconf==2.3.0
pettingzoo==1.24.1
pre-commit==3.5.0
redis==4.6.0
SuperSuit==3.9.0
termcolor==1.1.0
torch==2.0.1
tqdm==4.65.0
transformers==4.35.0
wandb==0.13.11

yasuohayashibara, 2023-12-05:
ありがとうございます．
agilerlを0.1.18から0.1.13に変更したら実行できるようになりました．

```
pip3 install agilerl==0.1.13
```

yasuohayashibara, 2023-12-05:
解決したのでissueを閉じます．

closed 2023-12-05 by yasuohayashibara